Naguissa/uRTCLib #2: int not write or read properly

hyperion11, 2017-12-26:
hello. My hands got to DS3231.
I try read and write to eeprom. Float and char work fine. Strange, but int - not.
void setup() {
  int inttmp = 32101;
  float floattmp = 3.14159;
  char chartmp = 'A';
if (!rtc.eeprom_write(0, inttmp)) {
    Serial.println("Failed to store INT");
  }
  if (!rtc.eeprom_write(4, floattmp)) {
    Serial.println("Failed to store FLOAT");
  }
  if (!rtc.eeprom_write(8, chartmp)) {
    Serial.println("Failed to store CHAR");
  }
nttmp = 0;
  floattmp = 0;
  chartmp = 0;
  Serial.print("int: ");
  rtc.eeprom_read(0, &inttmp);
  Serial.print(inttmp);
  Serial.print("; float: ");
  rtc.eeprom_read(4, &floattmp);
  Serial.print(floattmp);
  Serial.print("; char: ");
  rtc.eeprom_read(8, &chartmp);
  Serial.print(chartmp);
  Serial.println();

receive this
int: 65535; float: 3.14; char: A

hyperion11, 2017-12-26:
fixed by PR https://github.com/Naguissa/uRTCLib/pull/3

Naguissa, 2017-12-27:
Ues, maybe deleting the define and writing a note on README and examples is better option. I'll merge changes and update all ASAP, today was holiday in Catalonia.

Naguissa, 2017-12-27:
P.D.: I'll test on Maple, as I have a board here...

Naguissa, 2017-12-28:
Tested with new release, 4.0.2:
```
Serial OK
Board: STM32
int: 65535; float: 3.14; char: A
RTC DateTime: 15/5/2 16:42:19 DOW: 6 ---- 0: 101
RTC DateTime: 15/5/2 16:42:20 DOW: 6 ---- 1: 125
RTC DateTime: 15/5/2 16:42:21 DOW: 6 ---- 2: 0
RTC DateTime: 15/5/2 16:42:22 DOW: 6 ---- 3: 0
RTC DateTime: 15/5/2 16:42:23 DOW: 6 ---- 4: 208
RTC DateTime: 15/5/2 16:42:24 DOW: 6 ---- 5: 15
RTC DateTime: 15/5/2 16:42:25 DOW: 6 ---- 6: 73
RTC DateTime: 15/5/2 16:42:26 DOW: 6 ---- 7: 64
RTC DateTime: 15/5/2 16:42:27 DOW: 6 ---- 8: 65
RTC DateTime: 15/5/2 16:42:28 DOW: 6 ---- 9: 9
RTC DateTime: 15/5/2 16:42:29 DOW: 6 ---- 10: 10
RTC DateTime: 15/5/2 16:42:30 DOW: 6 ---- 11: 11
RTC DateTime: 15/5/2 16:42:31 DOW: 6 ---- 12: 12
RTC DateTime: 15/5/2 16:42:32 DOW: 6 ---- 13: 13
RTC DateTime: 15/5/2 16:42:33 DOW: 6 ---- 14: 14
RTC DateTime: 15/5/2 16:42:34 DOW: 6 ---- 15: 15
RTC DateTime: 15/5/2 16:42:35 DOW: 6 ---- 16: 16
RTC DateTime: 15/5/2 16:42:36 DOW: 6 ---- 17: 17
RTC DateTime: 15/5/2 16:42:37 DOW: 6 ---- 18: 18
RTC DateTime: 15/5/2 16:42:38 DOW: 6 ---- 19: 19
RTC DateTime: 15/5/2 16:42:39 DOW: 6 ---- 20: 20
RTC DateTime: 15/5/2 16:42:40 DOW: 6 ---- 21: 21
RTC DateTime: 15/5/2 16:42:41 DOW: 6 ---- 22: 22
RTC DateTime: 15/5/2 16:42:42 DOW: 6 ---- 23: 23
RTC DateTime: 15/5/2 16:42:43 DOW: 6 ---- 24: 24
RTC DateTime: 15/5/2 16:42:44 DOW: 6 ---- 25: 25
RTC DateTime: 15/5/2 16:42:45 DOW: 6 ---- 26: 26
RTC DateTime: 15/5/2 16:42:46 DOW: 6 ---- 27: 27
RTC DateTime: 15/5/2 16:42:47 DOW: 6 ---- 28: 28
RTC DateTime: 15/5/2 16:42:48 DOW: 6 ---- 29: 29
RTC DateTime: 15/5/2 16:42:49 DOW: 6 ---- 30: 30
RTC DateTime: 15/5/2 16:42:50 DOW: 6 ---- 31: 31
RTC DateTime: 15/5/2 16:42:51 DOW: 6 ---- 32: 32
```

If still failing recheck wiring and its length.

Also, I didn't integrate read/write in template to save code space at expense of one stack space; if you use multiple datatypes program space will grow too much if you do all procesing on template instead using the 1-byte function.

Naguissa, 2017-12-28:
Wow! Still failing on int!

Naguissa, 2017-12-30:
OK, see the problem:

On Arduino IDE default int size is 2 bytes: https://www.arduino.cc/reference/en/language/variables/data-types/int/

But when loading STM32 layer int size is changed to 4 bytes.

This leads to this warning:
```
Line 23: 	int inttmp = 0x00101010101010101010101010101010;

Compiling:

/tmp/arduino_modified_sketch_222052/uRTCLib_example.ino:23:15: warning: integer constant is too large for its type [enabled by default]
  int inttmp = 0x00101010101010101010101010101010;
               ^
/tmp/arduino_modified_sketch_222052/uRTCLib_example.ino: In function 'void setup()':
/tmp/arduino_modified_sketch_222052/uRTCLib_example.ino:23:15: warning: overflow in implicit constant conversion [-Woverflow]
```

Problem is that when running the sketch "sizeof(int)" gives 4, as it is in STM32.

The only fix I can find is to hardcode this case on STM32 (only), but still will fail in a direct assignment in sketch because compiler.

hyperion11, 2018-01-01:
wow. thank you! may be using int16_t \ uint16_t instead int may help..

Naguissa, 2018-01-04:
I accomplished to fix everything on STM32. I need to re-add Wire.begin fix to STM32 in order to 1st read don't fail. I've no idea why, but if I remove it program fails reading 1st EEPROM value.

Issue 

Still needs to be tested on ESP8266 and AVR, so I have not created a release.

Naguissa, 2018-01-17:
Solved on trunk for STM32, Ardiono AVR and Arduino SAM. Checking for ESP8266....

Naguissa, 2018-01-17:
OK! Tested on:

STM32
ESP8266
AVR
SAM

All works.

If yo have any trouble with wire.init() on library constructor you can call constructor with a FALSE on 1st parameter (all constructors) to skip it.

Released as 4.1.0: https://github.com/Naguissa/uRTCLib/releases/tag/4.1.0

closed 2018-01-17 by Naguissa