Naguissa/uRTCLib #5: Alarm exambles

tvixen, 2018-06-09:
Hi there
And thanks for the great code.

Would it be possible for you to make some alarm examples.

Thanks in advance.
Br Tvix

Naguissa, 2018-06-09:
Do you mean RTC's alarm or an alarm in a sketch?

tvixen, 2018-06-12:
I would appreciate if you could make a test sketch with some alarm examples.

tvixen, 2018-06-12:
If you have them in your lib ?  (Both alarm 1 and alarm 2)

Naguissa, 2018-06-12:
Hello.

About RTC's alarm, it's still not implemented. But remember that output is given by an RTC pin.

About alarm in sketch, it's an use case, it's not part of the library. It consists on checking the RTC value and trigger a pin when correct value is reached. Can be implemented in a lot of different ways inside sketch.

Naguissa, 2018-06-12:
Reading datasheets I'm checking:

**DS1307**: https://docs-emea.rs-online.com/webdocs/078a/0900766b8078aea1.pdf (page 8, bottom)

Its square output (4 modes) but must be controlled by microcontroller (you need to enable/disable it manually). The control could be added to library in the future.

**DS3231**: https://datasheets.maximintegrated.com/en/ds/DS3231.pdf (page 11, top)

Has two multimode alarms. This could be added to library in the future.

Also square output can be controlled manually, so also is a convenient thing to add to library.

maxholgasson, 2019-02-10:
Hi, is the alarm mode of the ds32312 SQW now supported? Im using the great uRTCLib but would need the alarm feature

Naguissa, 2019-02-12:
Hello,

It's still not implemented, I'll do it soon.

But, you know, it works using output pin from RTC, it's not an internal alarm (inside microcontroller).

It's quite easy so it should be available very soon.

Cheers.

maxholgasson, 2019-02-12:
Oh cool! Maybe this thread could ne helpful
https://forum.arduino.cc/index.php?topic=346940.0

When testing the function and using such a module there needs to be a resistor removed

BracketJohn, 2019-03-23:
Hi, are there any news on the alarm function? I love the premise of this lib being small, compared to all other libs for the DS modules, thanks for creating it!

Naguissa, 2019-03-25:
Unfortunately I have a lot of work now and also get hurt in my feet; I'll do ASAP, when I have few sparse time.

Naguissa, 2019-04-04:
I've created a new major release, 6.0.0: https://github.com/Naguissa/uRTCLib/releases/tag/6.0.0

New features are:

 - Added DS3232 datasheet and support (same as DS3231 but with SRAM)
 - Added Alarms support
 - Added SQWG support
 - Added SRAM support

You can see alarms usage on example uRTCLib_example_full.

See .h file to view different support and constants.

Cheers!

closed 2019-04-04 by Naguissa