Naguissa/uRTCLib #15: STM32 - (ST core) - %hhu format specifiers aren't supported.

prosper00, 2020-03-24:
Tested and got this working on an STM32F103 (bluepill) board this afternoon. Works great, however, in several of your examples, you use the %hhu format specifier, which isn't supported by newlib nano on the official ST arduino core for the STM32 (not sure about the roger clark/maple core). Simply using %u seems to work OK however.

Naguissa, 2020-03-25:
Hello,

Thanks for the report.

I've tested compiling it and doesn't give me any warning. I have not tested on the board, so I don't know if it fails there. Is it the case?

Cheers.

prosper00, 2020-03-25:
Yes, it compiles fine, however, the serial monitor literally prints "hu hu hu hu..."

Naguissa, 2020-03-27:
Fixed on new release, 6.2.5

Thanks!

closed 2020-03-27 by Naguissa