Naguissa/uRTCLib #16: Meaning of year as uint8_t

meltingSnowdrift, 2020-04-01:
The set() function and the year() function both use a uint8_t to represent a year value. However, the maximum value available in that datatype is lower than the current year. There does not appear to exist any explanation in the documentation regarding how years are represented as an 8-bit integer for either function.

In the code, the line in the `set` function that uses the "year" argument has a comment that simply says `// set year (0 to 99)`. Does this mean that the "year" values both when getting and setting the year correspond to the years 2000 to 2099? Can the documentation be changed to make it clear how years are represented in the library?

Naguissa, 2020-04-01:
If you see any of RTC datasheets, ie: https://github.com/Naguissa/uRTCLib_doc_and_extras/blob/master/extras/DS3232.pdf - page 11, look for 06h.

You can see that year value can only store 00 to 99 values.

I just realized, there's a century register (on 05h). But as it's for inform among 19xx and 20xx and it's quite old yet, I'll hardcode it to 1 on https://github.com/Naguissa/uRTCLib/blob/master/src/uRTCLib.cpp#L440 and provide a new release.

I'll also update and extend documentation.

This bit cannot be used to inform among 20xx and 21xx because it affects internally to leap years.

Thanks!

meltingSnowdrift, 2020-04-01:
Understood.

It may still be a good idea to allow the library to access the century register, as some users may want to use an RTC for applications using times in the past.

Additionally, it may in general be a good idea to make the year argument of the set and year functions go from 1900 to 2100 to make usage more intuitive. If this is done, the century and year registers can be both handled by the library, exposing only a year from 1900 to 2100 to the user for both setting and reading.

Naguissa, 2020-04-02:
Hello,

Century hardcoded to 1 on uRTCLib::set

It doesn't have any meaning to maintain 19xx support: we are 20 years ahead and the goal of RTCs is to maintain current date and time in a system. Leap years differ, so setting it wrong will lead to defective dates.

About years support, you can see it it the datasheets. i.e: https://github.com/Naguissa/uRTCLib_doc_and_extras/blob/master/extras/DS3232.pdf  - 1st page, at right. It states 1900 - 2099 years support.

I've created the release yet.

Thanks!

closed 2020-04-02 by Naguissa