Naguissa/uRTCLib #20: example for alarms

roblatour, 2020-05-17:
First, thank you for this library - appreciate the effort that you have put into it.

I see there was a closed issue already looking for an example of how to use the alarms.  You have noted that alarm support has been added, but the full example only shows how to set an alarm, not how to react to it.  Most importantly, how to set (or understand) the pin used to determine if the alarm is active or not.

adding a more detailed example would be very helpful.

Naguissa, 2020-05-28:
Added an example and created 6.2.9 release.

You can see connections here:
https://raw.githubusercontent.com/Naguissa/uRTCLib_doc_and_extras/master/imgs/deepsleep_connection.png 

You need a 10uF capacitor.

See added example (uRTCLib_deepsleep) to view details.

Currently tested on Arduino nano v3, but should work on ES8266 and ESP32.

Naguissa, 2020-05-28:
Oh! If you only want to detect the alarm it's quite easy:

SQWG outputs VCC on normal state and drops to GND when alarm is active.

You can connect it directly to any microcontroller pin.

closed 2020-05-28 by Naguissa