Naguissa/uRTCLib #24: ESP8266

JohnBelding, 2021-01-06:
I have been looking for your solution with the DS3231 for weeks and happily tried it.

I had to change a couple of things in your example code for the ESP8266, D1,D2 vs D3,D4 and DS3231 vs DS3232.

It compiled and uploaded fine, but did not work.  I ensured that I had put the Capacitor in correctly, but still didn't work.  This is what the uRTCLib_example running code provided:
Serial OK
RTC DateTime: 165/25/165 45:85:85 DOW: 165 - Temp: 0
RTC DateTime: 165/25/165 45:85:85 DOW: 165 - Temp: 0
RTC DateTime: 165/25/165 45:85:85 DOW: 165 - Temp: 0

I tried your other examples and got the same type of results.

I recompiled my original code (Makuna library) which blew up when I didn't have the capacitor installed and it worked fine and now resets the ESP8266 when the alarm is fired  -- didn't work before.

So, thanks for telling me about the capacitor, but I would like to get your code working too .

Any suggestions?

John B.

Naguissa, 2021-01-09:
Seems everything is wrong. I'll check on my own ASAP, but you could check GPIO pins meanwhile.

What editor / compiler do you use? I've observed problems on conditional defines on PlatformIO....

Naguissa, 2021-01-09:
You can check pin definitions using this table: https://www.foroelectro.net/electronica-digital-microcontroladores-f8/esp8266-equivalencia-entre-pines-y-gpio-t161.html

JohnBelding, 2021-01-09:
Pins are fine and work with a different library.

I'm using Arduino IDE with your library from github.

John B.

Naguissa, 2021-01-13:
Tested with LoLin v3 (ESP8266 ESP12E), "Example", adding `rtc.set_model(URTCLIB_MODEL_DS3231);` before `rtc.set` line, and everything works OK, without pullups or any extra component:

```

Serial OK
RTC DateTime: 15/5/2 16:42:0 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:1 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:2 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:3 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:4 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:5 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:6 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:7 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:8 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:9 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:10 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:11 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:12 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:13 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:14 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:15 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:16 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:17 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:18 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:19 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:20 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:21 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:22 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:23 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:24 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:25 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:26 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:27 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:28 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:29 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:30 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:31 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:32 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:33 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:34 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:35 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:36 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:37 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:38 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:39 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:40 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:41 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:42 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:43 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:44 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:45 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:46 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:47 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:48 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:49 DOW: 6 - Temp: 14
RTC DateTime: 15/5/2 16:42:50 DOW: 6 - Temp: 14
```
(date is wrong because I've used the example, which sets just that date-time).