Naguissa/uRTCLib #25: alarmDisable() only works with URTCLIB_ALARM_1

derbic7, 2022-03-30:
Hello, 
My question is that  alarmDisable() only works for URTCLIB_ALARM_1. I think the problem is in the following lines of code in uRTClib.cpp inside the mentioned function:
...
status &= 0b11111110;  // A1IE bit 
...

Which could be according to: 
..
status &= mask;
...

Also, Then I think you should also update the flag according to: 
_a2_mode = URTCLIB_ALARM_TYPE_2_NONE;

Thank you for your work and greetings

Naguissa, 2022-04-04:
It's just as you said; fixed and created a new 6.5.0 release.

I just become COVID positive last Friday, that's why I was so slow applying the changes....

closed 2022-04-04 by Naguissa

derbic7, 2022-04-06:
Thanks for the job.
If you allow me a suggestion, it would be nice to implement a function to read "Alarm Flag 'A1F' and 'A2F' bit" that allows checking if a certain alarm has been triggered, in case I don't want to use an interrupt or read the pin connected to INT/SQW. Something like, for example: 'alarmTriggered(const uint8_t alarm)' .
Thanks again and greetings.

Naguissa, 2022-04-06:
Thanks, it's a good idea. something like alarmStatus(alarm [1, 2 or any], refresh = true).

It seems we have new release soon....

Cheers!

Naguissa, 2022-04-08:
New release with that functionality:
https://github.com/Naguissa/uRTCLib/releases/tag/6.6.0

Also, 32K output pin functionality and performance improvements.