Naguissa/uRTCLib #26: deepsleep example alarm period

Grandpa-G, 2022-09-06:
I am trying to understand the example you have for deepsleep. It looks to me it should wake up every 20 seconds, however, it is waking up every minute. Can you explain the alarm a bit better?

In particular, how would I set up an alarm for 12 hours? I am using an Arduino d1 mini (WEMOS).

Naguissa, 2022-09-07:
Hello,

Alarms are triggered in a certain time, not X seconds/minutes after setting. Being so, if you uncommented rtc.set line it just sets RTC time to 0 seconds. As the next alarm is set to trigger each time it reaches "20" seconds, it will trigger after 20 seconds.

If you want to trigger after 12 hours you will need to read time, calculate new trigger time (now + 12h) and set alarm to that time and with mode URTCLIB_ALARM_TYPE_1_FIXED_HMS (Alarm 1 - Trigger every day at a fixed hour, minute and second).

Take in mind that these RTCs work as a wristwatch, you set the alarm not as "X hours from now", you set it at a fixed time.

And remember, comment-out rtc.set line after setting it to correct time.

Cheers!

Naguissa, 2022-09-07:
You can see alarm definitions on documentation, just here: https://naguissa.github.io/uRTCLib_doc_and_extras/uRTCLib_8h.html#a341fa3e12a7215c3350747e3e6bcb7ae

Grandpa-G, 2022-09-07:
Thanks for the explanation. I think that helps.

If I understand correctly, if I do this

```
 rtc.set(0, 0, 0, 1, 1, 6, 20);  // RTCLib::set(byte second, byte minute, byte hour, byte dayOfWeek, byte dayOfMonth, byte month, byte year)

  rtc.alarmSet(URTCLIB_ALARM_TYPE_1_FIXED_S, 20, 0, 0, 1); // Each minute, at just :20 seconds

```
and then deepsleep(0), I should get an alarm every 20 seconds since I reset the clock to 0 seconds and am asking for alarm at 20 seconds on the clock. Is this correct?

If so, it still doesn't work. This is a d1 mini so I have sqw connected to rst. What am I missing?

Naguissa, 2022-09-07:
Yes, you understood it perfectly.

From README:

Important notes

    Check .h file to see all constants and per-model limitations
    Alarm pin is normaly HIGH and turns LOW when active.
    When using alarms, you need to clear the alarm flag manually using alarmClearFlag(). If not done alarm maintains its LOW state.
    When using alarms SQWG is turned off. When using SQWG alarms are turned off. They're mutually excluding.

And the comments on the example says:

 *  **INPORTANT**: Place a 10uF capacitor between RTC's SQW pin and microcontroller input pin. If it's electrolytic, place - on microcontroller pin.
 *
 *
 * THIS SKETCH COMPATIBILITY LIST:
 *
 *  * ESP8266 - SQW ---> RST  
 *

So, if you are using the example I think you may miss the capacitor.....

Grandpa-G, 2022-09-07:
I have the capacitor.

Grandpa-G, 2022-09-07:
what does this mean?

`When using alarms SQWG is turned off. When using SQWG alarms are turned off. They're mutually excluding.`

Naguissa, 2022-09-07:
Alarm and SQWG (SQuare Wave Generator) uses same pin, so even internally to RTC they are mutually excludent.

On microcontroller, low must be only a pulse. On RTC ut should be low until you execute the clearAlarm function. That's why a capacitor is used; if microcontroller sees RST low will never start again.

On your 1st post it was working correctly, could you post your code here? It should have anything wrong....

Grandpa-G, 2022-09-07:
Maybe  you scared it. I just cleaned out code that wasn't relevant and 20 seconds worked. Now to try minutes, then hours.

Thanks, so far.

Grandpa-G, 2022-09-07:
I thought I had it figured out. Here is how I am setting my alarm. For this test, I am just trying to setting for 3 minutes in the future. When that works, I can change to set x hours in the future. I would have expected the alarm to trigger at 14:54, but nothing.

```
digitalClockDisplay();

uint8_t day = rtc.day();
 uint8_t hour = rtc.hour();
  uint8_t minute = rtc.minute();
  minute += 3;
  Serial.print("day: "); Serial.print(day); 
  Serial.print(" hour: "); Serial.print(hour); 
  Serial.print(" minute: "); Serial.println(minute);

  rtc.alarmSet(URTCLIB_ALARM_TYPE_2_FIXED_DHM, 0, minute, hour, day); // Each minute, at just :20 seconds
  // RTCLib::alarmSet(uint8_t type, uint8_t second, uint8_t minute, uint8_t hour, uint8_t day_dow);

Serial.println("Sleep forever");
  digitalClockDisplay();
  ESP.deepSleep(0); // 0 = sleep forever

```
The output is

```
15:15:02.818 -> 14:51:23 7 8 22
5:15:02.851 -> day: 7 hour: 14 minute: 54
15:15:02.851 -> Sleep forever
15:15:02.884 -> 14:51:23 7 8 22
```

Naguissa, 2022-09-08:
Could you attach whole file? 

1st thoughts:
 - May it miss one "rtc.refresh();" on top?
 - I think you still have the "rtc.set" line, because it shows same date/time on 2nd execution. If you want, you can change it to:

```
rtc.refresh();
if (rtc.lostPower()) {
		Serial.print("POWER FAILED. Clearing flag...");
		rtc.lostPowerClear();
		Serial.println(" done.");

		Serial.print("Setting new date due power lost... ");
		rtc.set(0, 0, 0, 1, 1, 6, 20);
		Serial.println(" done.");

	} else {
		Serial.println("POWER OK");
	}

```

Finaly, you will need any kind of auxiliary function to add minutes. Assuming globals: minute, hour, day:

```
void addMinutesToGlobals(uint8_t addMin) {
	minute += addMin;
	if (minute >= 60) {
		minute = minute % 60;
		hour += (uint8_t) (minute / 60);
		if (hour >= 24) {
			hour = hour % 24;
			day += (uint8_t) (hour / 24);
		}
	}
}

```

Naguissa, 2022-09-08:
Check RTC's battery. It should start as LIR. A lot of vendors sell them without rechargable battery (regular battery or not battery at all).

Even if it seems correct it's worth to try a new one, to see if it solves the problem.

Also, there're other problems with battery: https://forum.arduino.cc/t/ds3231-cr2032-vs-lir2032-warning-is-your-module-killing-the-battery/557346

Grandpa-G, 2022-09-16:
I have had to switch to a ESP32. I have recoded the alarm and sleep for ESP32. The processor goes into sleep, but immediately wakes up. I have set the time to be 2 minutes in the future for the test, yet it immediately wakes up. What am I doing wrong?

```
     uint8_t day = rtc.day();
      uint8_t hour = rtc.hour();
      //  hour += 1;
      uint8_t minute = rtc.minute();
      minute += 2;
      Serial.print("day: "); Serial.print(day);
      Serial.print("  "); Serial.print(hour);
      Serial.print(":"); Serial.println(minute);

      rtc.alarmSet(URTCLIB_ALARM_TYPE_2_FIXED_DHM, 0, minute, hour, day); // Each minute, at just :20 seconds
      // RTCLib::alarmSet(uint8_t type, uint8_t second, uint8_t minute, uint8_t hour, uint8_t day_dow);

      Serial.println("Sleep forever\n\n\n\n");
      digitalClockDisplay();
 
      esp_sleep_enable_ulp_wakeup();
      esp_deep_sleep_start();

```

Naguissa, 2023-08-09:
Very old issue. I had some busy days then and then forgot.

I close it....

closed 2023-08-09 by Naguissa