Naguissa/uRTCLib #27: Sketch stuck at rtc.enableBattery()

zekageri, 2022-10-03:
I'm using an ESP32-wrover-E ( 16mb flash 8mb psram )

After initializing the i2c and setting the address my sketch got stuck at `rtc.enableBattery()` and it never returns from that.

RTC is ds3231, using PlatformIO to code and latest from everything.

```cpp
#include "Arduino.h"
#include "uRTCLib.h"

uRTCLib rtc;

int rtcSDA_Pin  = 32;
int rtcSCL_Pin  = 33;

void setup(){
    Serial.begin(115200);
    vTaskDelay(1000);

    Serial.println("0");
    if( !URTCLIB_WIRE.begin(rtcSDA_Pin, rtcSCL_Pin) ){
        Serial.println("WIRE BEGIN FAILED");
    }
    Serial.println("1");
    rtc.set_rtc_address(DS3231_ADDR);
    Serial.println("2");
    rtc.set_model(URTCLIB_MODEL_DS3231);
    Serial.println("3");
    if (rtc.enableBattery()) {
        Serial.printf("[RTC] - Battery activated.\n");
    }else {
        Serial.printf("[RTC] - Battery activation failed.\n");
    }
    Serial.println("4");
}

void loop(){
}
```
4 is never printed on the Serial.

Naguissa, 2022-10-03:
Does it print any of:

 [RTC] - Battery activated.

or

[RTC] - Battery activation failed.

zekageri, 2022-10-03:
If i comment out `rtc.enableBattery()` it goes to the next line which is 

```cpp
isRTC_PowerLost = rtc.lostPower();
if ( isRTC_PowerLost ) {
    Serial.print("[RTC] - Power went down.\n");
    rtc.lostPowerClear();
} else {
    Serial.print("[RTC] - Power was ok.\n");
}
```

which prints `[RTC] - Power was ok.`

After i set a time with a wrapper func `rtc_setTime(50,35,20,1,3,10,21);`

```cpp
/*
*   Set the RTC time.
*/
void TimeSystem::rtc_setTime(byte sec,byte min, byte hour, byte dayOfWeek, byte dayOfMonth, byte month, byte year){
    rtc.set(sec, min, hour, dayOfWeek, dayOfMonth, month, year);
}
```

**And reading wrong value.**

```cpp
void TimeSystem::rtc_getTime(){
    rtc.refresh();
    year        = rtc.year();
    month       = rtc.month();
    dayOfMonth  = rtc.day();
    hour        = rtc.hour();
    min         = rtc.minute();
    sec         = rtc.second();
    dayOfWeek   = rtc.dayOfWeek();
    if( isDebugOn ){
        Serial.print("[RTC] - Time: %d %d/%d %d:%d:%d",year, month, dayOfMonth, hour, min, sec);
    }
}
```
**Which prints:** 
`[RTC] - Time: 165 25/165 45:85:857`

zekageri, 2022-10-03:
I2C scanner sees my ds3231

```cpp
[RTC] - Scanning I2C bus...
Found address: 104 (0x68)
Found 1 device(s).
```

But with 0x68 it does not pass trought this function.

arduinomnomnom, 2023-02-25:
Hi, I've just encountered this issue with a sketch that was working before. The ESP32 Dev board hangs at the enableBattery() line. Commenting it out and the sketch works.

Clue: I'm using the Arduino IDE 2.0.3 and **had just changed from board ESP32 v1.0.6 to v.2.0.7**

zekageri, 2023-02-25:
I have just switched to an other lib.

Naguissa, 2023-02-25:
I've to check it, maybe it's a timing issue, and would be good to add a I2C failure exit.

spaelectronics, 2023-03-27:
I am also having a problem enabling the battery on my ESP32-WROOM-32.

```
URTCLIB_WIRE.begin();
  myRTC.set_model(URTCLIB_MODEL_DS3231);

  if (myRTC.enableBattery()) {
    Serial.println("Battery activated correctly.");
  } else {
    Serial.println("ERROR activating battery.");
  }
```

I always get "ERROR activating battery." on the Serial Monitor.

But when I change code to disableBattery(), it works perfectly!

INFORMATION:

I'm using Arduino IDE 2.0.4, with version 1.0.6 ESP32 board set from https://github.com/espressif/arduino-esp32

spaelectronics, 2023-03-27:
As a matter of fact, rtc.lostPower() always returns false, even after a complete power failure (remove battery and remove vcc, and shorting battery pins to be sure)

The time is completely reset, but the lostPower() function still returns false.

I think this may be related to the same problem causing the enableBattery() function to fail.
Possibly an ESP32 compatibility issue?

Naguissa, 2023-03-27:
For sure! enableBattery should not be affected by refresh...

I have to recheck the library with my ESP32s and last arduino version....

Naguissa, 2023-03-27:
Oh, about RTC model, are you using  the DS3231 or DS3232?

spaelectronics, 2023-03-27:
I'm using DS3231

PR #35 (2024-01-27): More functionality and bug corrections
 Hello Naguissa,

Here is some more functionality and bug corrections:
- I2C communication blocking bug in enableBattery() and disableBattery() mostly hitting on ESP32 and sometimes on Atmega328p. enableBattery() not always working bug. Closes #27 
- addition of 12 hour / 24 hour clock mode select functionality
- flag to know if clock is in 12 hour / 24 hour mode and also AM-PM if in 12 hour mode
- added 12-24 hour mode select comment on example; a print statement that'll print AM-PM if in 12 hour mode
- model check before accessing SRAM or alarms, refres [... cut]
closed 2024-01-27 by Naguissa