Naguissa/uRTCLib #29: UNIX Timestamp

TCB13, 2023-01-16:
Hello,

Are there any plans to implement a method that will output a UNIX timestamp as an int? This would be very helpful since using timestamps is way easier than full dates.

Thank you.

Naguissa, 2023-01-17:
Not really. There're other mature projects for doing so in library manager, like TimeLib: https://github.com/PaulStoffregen/Time

closed 2023-01-17 by Naguissa

TCB13, 2023-01-17:
I was trying to avoid TimeLib because of its size. Having it builtin into uRTCLib was probably a few lines of code without much impact on this library.

Naguissa, 2023-01-17:
I've seen how Adafruit does it; do you think it could be interesting?

https://github.com/adafruit/RTClib/blob/master/src/RTClib.cpp

TCB13, 2023-01-18:
Hello,

If we're talking about https://github.com/adafruit/RTClib/blob/master/src/RTClib.cpp#L568 and https://github.com/adafruit/RTClib/blob/master/src/RTClib.cpp#L129 that's interesting.

Thank you.

Naguissa, 2023-01-23:
Done! https://github.com/Naguissa/uUnixDate

I'm requesting IDE insertion right now, may be in few days.

TCB13, 2023-01-24:
@Naguissa thank you for the new lib!