Naguissa/uRTCLib #32: no matching function for call to 'uRTCLib::set()

SzymenL0K0, 2023-09-08:
Hello. I try to have auto set time when i change battery and i use NTP to get time but i can't set the rtc with it

no matching function for call to 'uRTCLib::set(const uint8_t&, const uint8_t&, const uint8_t&, const uint8_t&, const uint8_t&, const uint8_t&)'

Trying something like this 
`
  const uint8_t year = 2000-timeClient.getYear();
  const uint8_t month = timeClient.getMonth();
  const uint8_t day = timeClient.getDay();
  const uint8_t hour = timeClient.getHours();
  const uint8_t minute = timeClient.getMinutes();
  const uint8_t second = timeClient.getSeconds();

    rtc.set(second, minute, hour, day, month, year);
    `
    (trying with byte too and directly and its not working too)

Naguissa, 2023-09-08:
Wich platform are you using? AVR, ESP, etc....

SzymenL0K0, 2023-09-08:
i write code using arduino IDE and my board is ESP32

SzymenL0K0, 2023-09-08:
When i try "static_cast<const uint8_t>(second)" 
and get "no matching function for call to 'uRTCLib::set(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)'" 
first time no& but still error :/

SzymenL0K0, 2023-09-08:
I'm really really sorry for bordering you. After like 3-4 hour i notice that i need to put what is day of the week ( like monday). In my function i have only 6 numbers instead of 7 :/

closed 2023-09-08 by SzymenL0K0

Naguissa, 2023-09-09:
Oh, it seems you are using set with less parameters than needed:

https://naguissa.github.io/uRTCLib_doc_and_extras/classuRTCLib.html#a85fe897310af5e33f986a54f2e3e32a5

void uRTCLib::set 	( 	const uint8_t  	second,
		const uint8_t  	minute,
		const uint8_t  	hour,
		const uint8_t  	dayOfWeek,
		const uint8_t  	dayOfMonth,
		const uint8_t  	month,
		const uint8_t  	year 
	)

Seems you missed dayOfWeek parameter...