Naguissa/uRTCLib #33: Why is SQW connected to RST, instead of the more usual pin 2?

samuelwhiskeyjohnson, 2024-01-21:
Just curious, for setting up the alarm,
why is SQW connected to RST, instead of the more usual pin 2 as interrupt pin?

Naguissa, 2024-01-21:
It's also possible. Done this way in order to simplify the example, as it runs on 3 types of MCUs with different capabilities.

closed 2024-01-21 by Naguissa

samuelwhiskeyjohnson, 2024-01-23:
Thanks for your reply
1. Are you saying I can use the deep sleep example by simply connecting to the pin 2? or is it hard coded in your library that users can ONLY use the RST pin?
[uRTCLib_deepsleep.ino](https://github.com/Naguissa/uRTCLib/blob/master/examples/uRTCLib_deepsleep/uRTCLib_deepsleep.ino)

2. Do I still need the10uF capacitor if I use pin 2? Or why use capacitor at all for RST pin too? I'm just curious because I've seen other RTC libraries just connect pin 2 to RTC directly without the capacitor.

Naguissa, 2024-01-23:
Also, if you want to use alarm again, remember to use alarmClearFlag() : https://naguissa.github.io/uRTCLib_doc_and_extras/classuRTCLib.html#a1832bbb5322fbe2895c14132e559e03a

samuelwhiskeyjohnson, 2024-01-27:
Hi thanks again for reply.
What do you mean by "The capacitor is used to 'release' RST pin, as MCU doesn't restart until released"?

At least for Arduino uno you don't need to release the RST pin? As long you send low signal for more than 2 micro seconds it restarts the Arduino.

Naguissa, 2024-01-27:
Hello again!

I reply in inverse order, as it's more logical:

2. RTC alarm toggles SQWG pin on it. It's how it works and it's useful to differentiate alarms and pulses generated by the generator. Hence the name (negated)INT / SQWG: https://github.com/Naguissa/uRTCLib_doc_and_extras/blob/master/extras/DS3231.pdf

1. That's why you need the capacitor on RESET pin, as MCU will never restart if RST is directly connected to SQWG, it will be always LOW. Also, in order to reuse alarms, you ALWAYS need to clearAlarm().