Naguissa/uRTCLib #38: Multi-byte read/write from RTC SRAM

BenUniqcode, 2024-02-21:
Hi
First of all thanks for this library, and for so promptly incorporating my earlier PRs. 
This time around I have a suggestion but don't have a PR because I'm not sure how best to do it.
In Adafruit RTCLib there are functions to read and write an arbitrary number of consecutive bytes from/to the RTC's SRAM into a buffer in one go. In fact the 1-byte read and write just call these functions with number of bytes = 1.
It would be great to have this in uRTCLib too. I am currently working around it by reading one byte at a time, but that obviously comes with additional I2C overhead which it would be good to avoid. 
Cheers
Ben

Naguissa, 2024-02-21:
I can do it as uEEPROMLib, but then I would rather do it in a spare library as it can use too much resources.

BenUniqcode, 2024-02-21:
It shouldn't be necessary to break it out into a separate library, it's just 2 small functions/overrides, I don't think it seriously threatens the "micro" nature of the lib ;) I've looked into it a bit more and I think I can probably figure it out, so I'll send you a PR when I get some time.

Naguissa, 2024-02-21:
Problem is that using templates (on .h file, at end) can lead to a massive code explosion. Also, for paged reads you need at least 32 bytes buffer.

BenUniqcode, 2024-02-24:
I'm literally just after something like
int uRTCLib::ramRead(const uint8_t address, byte *buffer, uint8_t length);
and similar for write. I don't see how that would cause a code explosion.
I don't care about supporting read/write of arbitrary types; this is just about reading/writing a series of bytes all at once with less I2C overhead than doing one at a time. These RTCs supports reading or writing an arbitrary number of consecutive bytes, so I don't know why a 32-byte buffer would be needed?

Naguissa, 2024-09-11:
I was thinking about this and the I realized: Why duplicate the code if you can use uEEPROMLib directly?

```
#include "uEEPROMLib.h"

// ...

#define RTC_RAM_READ_DS1307(pos, len)  eepromRTC.eeprom_read(0x08 + pos, (byte *) c_string, len)
#define RTC_RAM_READ_DS3232(pos, len)  eepromRTC.eeprom_read(0x14 + pos, (byte *) c_string, len)

uEEPROMLib eepromRTC(0x68); // Use RTC address

```

Then you can use it directly.

Naguissa, 2024-09-11:
For write functions:
```
#define RTC_RAM_WRITE_DS1307(pos, ptr, len)  eepromRTC.eeprom_write(0x08 + pos, ptr, len)
#define RTC_RAM_WRITE_DS3232(pos, ptr, len)  eepromRTC.eeprom_write(0x14 + pos, ptr, len)
```

closed 2026-02-05 by Naguissa