Naguissa/uRTCLib #39: Trying to use alarm to wake up from sleep. But when RST pin is connected I get AVRdude upload error.

samuelwhiskeyjohnson, 2024-03-03:
Using arduino UNO
I'm getting this error when I connect SQW pin on RTC (with capacitor in between like you instructed in https://raw.githubusercontent.com/Naguissa/uRTCLib_doc_and_extras/master/imgs/deepsleep_connection.png) with RESET pin on Arduino.

```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
...
Failed uploading: uploading error: exit status 1
```

Basically the issue comes down to when RST pin is either held high or held low, it creates this error.
The issue gets fixed when RST pin is not connected to anything. 

Any idea on how to fix?

Naguissa, 2024-03-03:
In short, remove capacitor and RTC from RESET pin while programming.

Detailed explanation: When programming MCU, 1st step is reset MCU and thanks to its bootloader, connect it in programming mode. As you have a capacitor and the RTC connected to the MCU, they are interfering with typical process and fooling the MCU and the programmer. Disconnecting them to program the MCU you'll solve the problem.

Naguissa, 2024-03-03:
P.S.: I don't know your MCU, but a lot of them can operate with wake-up pin interrupts. Using RESET to do so is the most compatible way, but if your MCU supports it maybe using a pin-interrupt is way more beneficial....

samuelwhiskeyjohnson, 2024-03-10:
It worked! I used pin 2 as interrupt pin and [attachInterrupt()](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/).

Do you still need capacitor between SQW pin on RTC and pin 2 (interrupt pin) on Arduino?

Naguissa, 2024-03-10:
Perfect!

Using any other pin than RST does not need any capacitor.

closed 2024-03-10 by Naguissa