Naguissa/uRTCLib #43: MCU With More Than One I2C

mohdrais, 2024-08-05:
Hello, I am using Pico W and I want to run DS3231 by using I2C1 (Wire1). How to do that? (It works when using I2C0 (Wire).

Naguissa, 2024-08-05:
I'll try to define it as constant and try to be able to define it before the include. But last time I tried it didn't work (it was like 2+ years ago)

Naguissa, 2024-08-06:
I've seen it's already done for Attiny. You can test it definining URTCLIB_WIRE before including the library:

```
	#define URTCLIB_WIRE Wire1
	#include "uRTCLib.h"
```

mohdrais, 2024-08-07:
Sorry, it does not work for Pico W. Thank you

Naguissa, 2024-08-07:
Then the quick way is to copy library on Sketch's folder (under libraries folder) and change definition line:

https://github.com/Naguissa/uRTCLib/blob/master/src/uRTCLib.h#L48

Change it from Wire to Wire1

So, this way folders will be like:

```
[...]/Arduino/YourSketch/libraries/uRTCLib
[...]/Arduino/YourSketch/libraries/uRTCLib/src
[...]/Arduino/YourSketch/libraries/uRTCLib/src/uRTCLib.h ==> This is the modified file
[...]/Arduino/YourSketch/libraries/uRTCLib/src/uRTCLib.cpp

```

closed 2024-08-08 by Naguissa